ToddHuaJi/FlappyDrone #12: Drafting block diagram

ToddHuaJi, 2018-09-19:
Convert picture on Slack into a real diagram.

ToddHuaJi, 2018-09-19:
[image]

ConleyGao, 2018-09-19:
This is our first draft what you think, if you think this is perfect ,comment " perfect" otherwise leave suggestion, once we have finalize the design, we will ask prof Zink for last review

tony5598, 2018-09-20:
I'm not sure if this would be correct but I would encompass both the data processing box and the data collection/motor box into a bigger box indicating that those two portions are on the actual drone itself

The UI is on the controller

ToddHuaJi, 2018-09-20:
Data Processing block definitely can be more detailed, but that won't happen until we figure out what we are going to do with it. Just keeping it on the high level for now.

ConleyGao, 2018-09-20:
> I'm not sure if this would be correct but I would encompass both the data processing box and the data collection/motor box into a bigger box indicating that those two portions are on the actual drone itself
> 
> The UI is on the controller

These blocks are separated by functionality, We discard the Idea of describing the system physically , we still don't have the hardware configuration, for example ,sensors, there are build-in sensors  and external sensors , we don't know many sensors are installed on drone. 
as todd said it can be more detailed , once we know the setup.

ToddHuaJi, 2018-09-20:
> I'm not sure if this would be correct but I would encompass both the data processing box and the data collection/motor box into a bigger box indicating that those two portions are on the actual drone itself
> 
> The UI is on the controller

- We are trying to separate hardware and software. We might want to take the Raspebery pi block away, because this is kind of redundant here. I am acutally thinking replacing that with "Power Management" to match power supply on the left. What do you think about this? @ConleyGao 

- We also have the alarm which may or may not be on the controller. UI will be flexible, so I would suggest keep an open mind here. We might get creative if we have some free time towards the end.

tony5598, 2018-09-20:
The Alarm is on the controller, I'll post a link to the controller on the specs issue page

closed 2018-09-20 by kaiqiann

ToddHuaJi, 2018-09-20:
> Since motor control is giving feedback to the motor, I think the arrow between Data Collection and Motor and Data Processing will be enough. I think motor control is not necessary, the arrow will indicate the interaction.

Motor control does not simply give feedback to the motor. It is also in charge of ignoring user control when the drone is too close to a surface. Therefore, motor control will stay.

kaiqiann, 2018-09-20:
shouldn't add interaction between data collection and motor block to UI block? since I assume the feature switch is to trigger sensors on/off
or you just directly control the data analysis on/off while the sensors are still generating data?