ToddHuaJi/FlappyDrone #13: Problem statement

ToddHuaJi, 2018-09-19:
For bench side meeting

tony5598, 2018-09-20:
Here are some bullet points for the problem statement, I tried to word it in layman's terms: 
•	While drones are very agile aircraft, flying them indoors still poses many risks.
•	Even experienced pilots will have issues avoiding objects, obstacles, and even simply flying low enough as to not hit the ceiling or floor
•	Therefore, drone pilots need a sophisticated object avoidance system that will aid in indoor flight while at the same time not hinder the drone’s function
•	This will allow for efficient but safe indoor function of the drone

ToddHuaJi, 2018-09-20:
I think it's good, and I think we can add this:
- This will also improve flight safety for autonomous drones.

ConleyGao, 2018-09-20:
> I think it's good, and I think we can add this:
> 
> * This will also improve flight safety for autonomous drones.

This is similar to last tony's last bullet , maybe we can merge to something like 
•	This will improve flight safety and provide the efficiency for autonomous drones.

kaiqiann, 2018-09-20:
I am on board. it's good. When I write down the problem statement earlier, I mentioned that drone is relatively expensive. So, you could add like the system could reduce company's expenses in some way, something like that.  should I close this issue/???