ToddHuaJi/FlappyDrone #14: Assesment

ToddHuaJi, 2018-09-19:
For bench side meeting.

tony5598, 2018-09-20:
http://www.spektrumrc.com/Products/Default.aspx?ProdId=SPM8000

tony5598, 2018-09-20:
Here is the manual for the controller as well: http://www.spektrumrc.com/ProdInfo/Files/SPM8000_Manual_EN.pdf

tony5598, 2018-09-20:
One of the features of the controller is programmable voice alerts, so I think that's what they mean when they say alarm

ToddHuaJi, 2018-09-20:
> One of the features of the controller is programmable voice alerts, so I think that's what they mean when they say alarm

That solves the alarm thing. Is there any free switch on the controller we can use to control turn our sensors on/off?

kaiqiann, 2018-09-20:
Specification:
1.	The pilot in control uses the controller to initialize the flight and control the drone to move
2.	Sensors placed on the drone are collecting data as the flight was initialized
3.	Detects surfaces left, right, forward, behind, above, and below from the drone outer body perimeter while in stable flight and determines distances left, right, forward, behind, above and below from the drone outer body perimeter
4.	Compared the distance that calculated with the adjustable outer avoidance offset distance (which is to be determined) to trigger the audible alarms in the stable flight
5.	Compared the distance that calculated with the adjustable inner avoidance offset distance (which is to be determined) to prohibit the collision and keep drone airborne
6.	Pilot in control is noticed and adjusts the drone’s position to resume the task

ToddHuaJi, 2018-09-20:
Can you rework the sentences a bit? They just look really really awkward, but I get what you are saying.
I am ok with a reworked version

kaiqiann, 2018-09-20:
that is what it has written on the spec. Sure. I can rephrase it.

kaiqiann, 2018-09-20:
Specification:
1.	The pilot in control uses the controller to initialize the flight and control the drone to move
2.	Sensors placed on the drone are collecting data as the flight was initialized
3.	Sensors detect and determine distances in all directions (the distances are left, right, forward, behind, above and below from the drone outer body perimeter)
4.	The data analysis will compare the calculated distance with the inner and outer avoidance offset distance to trigger the audible alarms to notice the pilot or to prohibit the collision and keep drone airborne
5.	Pilot in control is noticed and adjusts the drone’s position to resume the task

ToddHuaJi, 2018-09-20:
I am ok with this version

ConleyGao, 2018-09-20:
anything about power consumption ?

ConleyGao, 2018-09-20:
I think this is good

ToddHuaJi, 2018-09-22:
This ticket is blocking #17

tony5598, 2018-09-22:
I'm ok with these specs

closed 2018-09-22 by tony5598