ToddHuaJi/FlappyDrone #17: Draft up a problem statement/specification/block diagram initial draft for our benchside meeting

ToddHuaJi, 2018-09-20:
Combine #14 #12 #13 into one sheet of paper, double-sided.

This can start once all three issues are reviewed and closed.

ToddHuaJi, 2018-09-20:
- [ ] #14  

- [x] #12  

- [x] #13

ToddHuaJi, 2018-09-22:
All blocking issues were closed. This issue is unblocked

ToddHuaJi, 2018-09-22:
Here is the link for our first daft: [https://docs.google.com/document/d/10ubwM2a-3D57OoUZ7GkltBiqKIZPfpf0nDRbTAOUK24/edit](url)

Take a look and let me know if there is anything you would like to change.

kaiqiann, 2018-09-24:
I am ok with the draft

ToddHuaJi, 2018-09-30:
After today's meeting with Wally. I think we need to rework our problem statement and system spec.

- [ ] I will add the humidity part to it. 

- [ ] We probably need a better block diagram as well.

ToddHuaJi, 2018-10-01:
Closing this issue, because it is reviewed and due tomorrow

closed 2018-10-01 by ToddHuaJi