ToddHuaJi/FlappyDrone #18: design spring review format for weekly meeting

ConleyGao, 2018-09-20:
Since each meeting is time limited , I want to setup a structure for the meeting so we do not wonder off to random stuff and waste time. This is what I have now, looking for feedback ... 

As project manager, I would like to received feedback from team member for each sprint

Topic 1 ,feedback on amount of Work assigned in this sprint
why  did you not finish, do you think you should get more tasks or less ?? 

Topic 2, Evaluation of unfinished work ( carry over ? or  set aside ) 

Topic 3, new issues , open discussion about new issues needed for the new sprint 

Topic 4 , Each member state like and dislike about this sprint(can be anything )  , so we do not accumulate bad feelings

ToddHuaJi, 2018-09-20:
I think the overall structure is good, we can try it out on Monday. Here are something I want to suggest: 

- I want to suggest we give it a time limit for each topic. We only have 1 hour meeting time, so I suggest the following:
Topic 1+2, 30min
Topic 3, 15min
Topic 4, 5min
We have 10 min for the announcements and buffer time in case we run over. **These are most likely to change.** 
- We should take turn to be note taker in the meetings. Just record in humanly readable format. We will need those as part of the record.

ConleyGao, 2018-09-22:
closing this , no feedback from other member , this is finalized with my plan + todd's suggestion

closed 2018-09-22 by ConleyGao