ToddHuaJi/FlappyDrone #20: coordinate a first meeting with Mr. Sayed

ConleyGao, 2018-09-22:
He's available  8a-8p. (774) 217-1866
-When do we want to make a call, 
-List of question ?
-who want to contact Sayed, or  a group called with everyone around just in case of new question pop up?? 
-plan for a day to meet him in UMass

ConleyGao, 2018-09-22:
I can call him maybe tomorrow but I need answers for these questions

ToddHuaJi, 2018-09-22:
I suggest you text him first. 
Aske him does he want to call us individually or a group call. I would suggest noting heavy first

ConleyGao, 2018-09-22:
Texted, waiting for response

ConleyGao, 2018-09-23:
group meeting is preferred , we need to setup a time

ToddHuaJi, 2018-09-24:
Let's talk about this in today's meeting. We should have enough time.

ConleyGao, 2018-09-24:
please generate a list of questions You would like to ask . 
1. what parts would be sponsored by them 
2. when would we start getting parts , such as drone kit controller 
3. who is our consultant, do we reach him ?? 
4. is there more specific requirements, 
5. any restriction on open source code

ToddHuaJi, 2018-09-24:
I have another question:

Is there any special restriction on how much weight we can mount on the drone? Are we free to mount whatever we need with the weight capacity of the drone?

I am asking this because we might need to use a motor to rotate the depth vision camera, which might add a lot of weight.

ToddHuaJi, 2018-09-24:
I have more questions, so I am going to make a list of what we have 

1. what parts would be sponsored by them
2. when would we start getting parts, such as drone kit controller
3. who is our consultant, do we reach him ??
4. is there more specific requirements,
5. any restriction on open source code
6. Is there any special restriction on how much weight we can mount on the drone? Are we free to mount whatever we need with the weight capacity of the drone?
7. What kind of obstacle we are talking about here? A surface like a wall? What's the smallest obstacle we need to worry about?

ToddHuaJi, 2018-09-25:
1.ask them to come to our review
2. do they want weekly report?

ToddHuaJi, 2018-09-27:
Did that

closed 2018-09-27 by ToddHuaJi