ToddHuaJi/FlappyDrone #23: Start to prepare PDR presentation

kaiqiann, 2018-09-24:


kaiqiann, 2018-09-24:
- consider deliverable issues after meeting with Sayed
- meet on Saturday

ToddHuaJi, 2018-10-01:
Here is the link to our PDR:https://docs.google.com/presentation/d/1ke-dhvkZCmDU4zilhXsWPssDaMqAtWmpDntq2N7tsnU/edit?usp=sharing
I will start putting stuff that we all agree on to there.

ToddHuaJi, 2018-10-03:
Worked out some major design issue with the team today.
**Something else we can do better: pictures!**

ToddHuaJi, 2018-10-11:
Some feedback from Prof. Zink:

- [ ] 1. Slide 5, Change the title to something else 
- [ ] 2. Slide 6. Change “offset” into safety distance.
- [ ] 3. Slide 8, Change “Object analysis” into “Object detection”.
- [ ] 4. Ultrasound sensor might be sensitive to land environment
- [ ] 5. LiDAR might not be reliable in bright environment 
- [ ] 6. Echo with the ultrasound might be reason to dismiss ultrasound
- [ ] 7. Put some numbers and ranges in PDR slides
- [ ] 8. Simplify “Nice to have” slide, that might be too ambitious. Scale back a little it on this
- [ ] 9. Mention the budget issue in the PDR slide

ToddHuaJi, 2018-10-16:
Finished the PDR presentation, waiting for final review report.