ToddHuaJi/FlappyDrone #32: Software system plan

ConleyGao, 2018-09-27:


tony5598, 2018-09-29:
So I realized that before I can plan out the data flow of our system, it'd be a good a idea to research a couple different approaches to how we can process our data and which method we'll use to collect our data (this sort of overlaps with algorithmic research). I found three interesting papers on this topic. The first 2 papers deal with algorithmic approaches to object detection. These papers, however, examine dynamic flying environments (an environment with multiple moving objects), we won't have to deal with this issue thankfully. Both papers mention A* search which I am particular interested in. This would involve taking the live video feed we'd be receiving and turning it into a grid that A* would operate on. Grids that have objects in them would be given an infinite weight, letting the drone no that it cannot go there. The third paper I looked at, examines applying Convolutional Neural Networks to perform object detection. Considering that this approach would require heavy processing, we might be able to interface an off board computing ASIC to do these calculations faster (this could end up being the PCB portion of our project). We could use an old GPU in combination with a PCB to do this although I'm not well versed on such an approach. Here are links to the papers: 

https://etd.auburn.edu/bitstream/handle/10415/2976/holtThesis.pdf?sequence=2&ts=1438796208696

https://etd.auburn.edu/bitstream/handle/10415/3019/masterthesis.pdf?sequence=2&ts=1429155373562

http://dsc.soic.indiana.edu/publications/cloud_robotics_IEEE_Robotic_Computing_2017.pdf

ToddHuaJi, 2018-09-29:
About that A* search, that's more of a machine learning problem and finding the solution. We kind of skip that part, but it might be useful.

That CNN does sound interesting. If we are going to go with CNN, we actually can use some open source computer vision library which is an implementation of this idea. I believe there is research that focuses on the vision side, especially in the stereoscopic part. I came across a few last semesters.

ToddHuaJi, 2018-10-03:
[image]
Here is a rough draft of our agreed block diagram