ToddHuaJi/FlappyDrone #34: sensor research

ConleyGao, 2018-09-27:


kaiqiann, 2018-09-29:
[Sensors research.docx](https://github.com/ToddHuaJi/FlappyDrone/files/2430130/Sensors.research.docx)

I have added one or two sensors in each different categories. let me know if you guys need more information

ToddHuaJi, 2018-09-29:
**Can you upload this to our google drive? I need to make notes and comments there.**

FYI, that stereoscopic camera software is PC only. I would say find a few more stereoscopic cameras because that might be the main detection sensor. The other ones look okay though.

ConleyGao, 2018-10-01:
http://ardupilot.org/copter/docs/common-lightware-sf40c-objectavoidance.html#common-lightware-sf40c-objectavoidance

http://ardupilot.org/copter/docs/common-teraranger-tower-objectavoidance.html#common-teraranger-tower-objectavoidance

Could you take a look at these two sensors , are they suitable for this project ?? power consumption ... things like that  so we can decide which one is better.
Also, for the sensors you found , do they use MAVLink protocols ?

ToddHuaJi, 2018-10-01:
> http://ardupilot.org/copter/docs/common-lightware-sf40c-objectavoidance.html#common-lightware-sf40c-objectavoidance [...]

The first one cost $999, [https://lightware.co.za/products/sf40-c-100-m](url) If one motor is 10W, which is unlikely, then without considering the Pi, we have 40+20 = 60 W. This will drain our battery pretty fast.
The sensor itself will spin, and that can be a pain for our gyrometer. We need to compensate the counter spin from drone, and this will drain the battery even more! 

The second one is $699, [https://www.robotshop.com/en/terabee-teraranger-tower-evo-600hz-8-sensors-tof-rangefinder.html?gclid=Cj0KCQjw6MHdBRCtARIsAEigMxHE-s-IiIubSlqKmcy-Vp176OPiTDFzRt38yGI-p0tbW3DFeAIxuoQaAkjAEALw_wcB](url). The power consumption is less, and it looks very easy to mount. The only thing I am worried about this thing is the **accuracy**. **±12cm** is quite awful for our close range detection. You can miss almost 5 inches with this thing.

closed 2018-10-15 by ConleyGao