ToddHuaJi/FlappyDrone #41: Ordering One sensor for testing purpous

ToddHuaJi, 2018-10-10:
How do you guys feel about ordering one sensor this week so it will get here right after PDR. We can start looking at how hard it is to implement our own sensor drive(FrontEnd and BackEnd)

ToddHuaJi, 2018-10-10:
I personally think there is no blocking issue on this. We should start messing with ArduPilot library soon.

kaiqiann, 2018-10-10:
it's probably the thing we will do right after PDR. And yes. I suggest we could order the Plan A sensor, which costs 40 bucks for one.

ToddHuaJi, 2018-10-10:
> it's probably the thing we will do right after PDR. And yes. I suggest we could order the Plan A sensor, which costs 40 bucks for one.

Can you fill out a bill of material for Mr.Sayed to order the purchase? There should be a template on the SDP desktop. Please bring this up in the meeting tomorrow.

ToddHuaJi, 2018-10-16:
Closing this because we gave Wally the ordering sheet.

closed 2018-10-16 by ToddHuaJi