ToddHuaJi/FlappyDrone #42: Verify OS, Configure OS with PI and Navio

ConleyGao, 2018-10-15:
very, set configuration

ConleyGao, 2018-10-16:
1. setup the keyboard config to proper us Key board 
2. updated OS

ConleyGao, 2018-10-16:
3. updated Navio

ConleyGao, 2018-10-16:
pi itself is fine , Someone else do some configuration with the Ardupilot ex GCS IP,  bluetooth

ToddHuaJi, 2018-10-16:
> pi itself is fine , Someone else do some configuration with the Ardupilot ex GCS IP, bluetooth

which step is it on the installation guide?

ToddHuaJi, 2018-10-22:
We need a simple router, in this case a cellphone hotspot would be perfert to connect.

*Second attempt on connecting with Mission Planner(MP)** 
Connecting the pi with ethernet cable failed
I used my cell phone as hotspot and connected my laptop and the pi to my hotspot network. I did change the ip address in the process. This process takes a couple of minutes and it is irritating, but it works

**Exploring the possibility of switching wifi**
Since we don’t have any GUI, the best I can do now is the following:

1. Turn on my hotspot
2. reboot/turn on the pi
3. The pi should automatically connect to the hotspot 
4. Check for IP address
5. Start connection

To switch back to eduroam:

- Turn of hotspot
- Reboot

ToddHuaJi, 2018-10-29:
We are facing two problems:
1. The Pi/Navio is confused about which stick is throttle.
2. ESC is configured yet.

ToddHuaJi, 2018-11-06:
Closing, redefined in #60

closed 2018-11-06 by ToddHuaJi