ToddHuaJi/FlappyDrone #44: Configure the Wifi on PI

ConleyGao, 2018-10-15:
- make sure the wifi works on pi

ConleyGao, 2018-10-16:
SecureW2 is installed but not working  return " Could not configure wireless network "
Maybe helpful  https://www.reddit.com/r/raspberry_pi/comments/3r7pwh/connecting_to_eduroam_issues/

chatted with UMASS IT , they told me SecureW2 does not support raspian .  They might able to help if we bring the device to them ( I don't trust them ) .

ToddHuaJi, 2018-10-16:
> SecureW2 is installed but not working return " Could not configure wireless network "
> Maybe helpful https://www.reddit.com/r/raspberry_pi/comments/3r7pwh/connecting_to_eduroam_issues/
>
> chatted with [...]

- I remember there is a way to manually set up the wifi. You have to select which authentication and which 2nd layer authentication. I am not sure whether we can use that or not, but the instruction can be found at the bottom of this page: https://www.umass.edu/it/eduroam-setup-guides
- You mentioned it is working with the ethernet cable. Can we just plug it in all the time? We don't exactly need the desktop ;)

ConleyGao, 2018-10-16:
no sure how to do bullet one 

ethernet cable is a solution but wifi will be better because we can ssh into it when it's on flight

ToddHuaJi, 2018-10-16:
I don't think it would be safe to ssh into it mid-flight. I think we can get by with ethernet for now, unless something else absolutely needs WiFi. The Pi shouldn't leave the lab unless we are testing it out anyway.

ToddHuaJi, 2018-10-18:
This might be a problem when we are trying to connect with mission planner.

ToddHuaJi, 2018-10-18:
All praise to https://gist.github.com/AlexLamson/b6812a4e0c3434901750661514a6f9fc 

Confirmed no ethernet connection.
I also reboot the pi to make sure #45 is still working, and I can successfully ssh in!

Marking read for review.

ToddHuaJi, 2018-10-21:
Due to an unfortunate event, we reflashed the system, but wifi still works following above instruction.
Closing this because it is verified.

closed 2018-10-21 by ToddHuaJi

ConleyGao, 2018-10-21:
Could you be specific about unfortunate event so the same problem won't happen again

ToddHuaJi, 2018-10-21:
> Could you be specific about unfortunate event so the same problem won't happen again

Will post work log once finished