ToddHuaJi/FlappyDrone #45: SSH Setup

ConleyGao, 2018-10-15:
options 
block by wifi issue 
- NGrok 
- computer service

ToddHuaJi, 2018-10-16:
Blocked by #44

ToddHuaJi, 2018-10-16:
> Go to https://dashboard.ngrok.com/status you will see "tcp://0.tcp.ngrok.io:15888" something like this
> website id " [hokaiimpact2@gmail.com](mailto:hokaiimpact2@gmail.com)" , pw: " floppybird"
> For this example, the host name is 0.tcp.ngrok.io , the port is 15888
> Then you know how to ssh
> 
> To know 👍
> The process execute by crontab at boot
> port number changes ( because I use free account )

Can you write down the instruction of how to set up? Where can we find the open SSH port?
Also,
 > website id " [hokaiimpact2@gmail.com](mailto:hokaiimpact2@gmail.com)" , pw: " floppybird"

 can' t access  https://dashboard.ngrok.com/status