ToddHuaJi/FlappyDrone #57: [MDR Deliverables] ---- Giving out warning signals when Detecting a surface

ToddHuaJi, 2018-11-06:
**Feel free to create sub-ticket for this task and link that ticket to his one**

Lead: Tony

Developing standalone Cpp code for the following:
- [ ] Giving out **command line** warning when sensor reading(individual sensor) is below a certain number (Distance Limit)

- [ ] Being able to read a **custom** int for Distance Limit from a different file (text?) 

- [ ] [**Bonus for MDR**]  Having **two different Distance Limit**, warning and suspend fligh. The code is able to read them from a specific file.

- [ ] [**Bonus for MDR**] Being able to map the warning to a audio alarm on the RC control (Pilot control on the ground)

ToddHuaJi, 2018-11-06:
Related issue: #25 #53

ToddHuaJi, 2018-11-09:
Uploaded a feature that satisfies the first three item of this ticket.

Waiting for review. @kaiqiann @ConleyGao @tony5598

kaiqiann, 2019-01-28:
MDR deliverable is done

closed 2019-01-28 by kaiqiann