ToddHuaJi/FlappyDrone #63: [MDR Presentation] ---- MDR slides, documents, delivererables

ToddHuaJi, 2018-11-07:
Deliverables:

- [ ] #62 

- [ ] #57 

- [ ] #60

ToddHuaJi, 2018-11-10:
Here is our 5 deliverables:
1. Assemble, configure, initialize flight control system (Navio on Pi)
2. Get reading in cm from at least TFmini LiDAR with the accuracy specified on the datasheet (1%(less than 6m), 2%(6-12m))
3. Being able to detect a surface within sensor effective range(0.3-12m) and send a warning message to the console when the reading is smaller than the safety distance (user specified)
4. Assemble and calibrate the drone so that the drone is ready to fly with original firmware when needed.
5. Proof of concept on non-compass based orientation. Proof includes needed data and algorithms.

For the MDR presentation, we need to name who is charge of which deliverable. I am proposing the following:
@ToddHuaJi @kaiqiann : We take the first three, because these are related.
@ConleyGao : Number 4
@tony5598 : Number 5 

This is who is demoing which delieverable. Each person is in charge of coming up sufficient way to demo how it is finished.

For example, we plan to make a video of sensor giving out different reading compared to different distance.

ToddHuaJi, 2018-11-14:
I started the MDR slide https://docs.google.com/presentation/d/1f1uv3jaEXOQ44WSXIKjHXSR098LEUj26CU7uPAoMoSU/edit?usp=sharing

kaiqiann, 2019-01-28:
MDR has gone. prepare for CDR

closed 2019-01-28 by kaiqiann